just-trey/ankermake-m5-profile #4: What about SuperSlicer

AutismeLuxembourg, 2022-12-25:
the Profile works fine in PrusaSlicer, but SuperSlicer creates an error on Startup: Preset bundle `AnkerMake` version not found in index: 0.1.0

LazeMSS, 2022-12-28:
Works fine for me :)

just-trey, 2022-12-28:
I was able to get it loaded as well. You may have a corrupt file somewhere. did you put it in the correct folder? steps are the same for Prusa Slicer and Super Slicer (just open Super Slicer to show the right configuration folder) 

https://github.com/just-trey/AnkerMake-M5-Profile/tree/master/PrusaSlicer-2.5-M5-Profile#installing-the-profile

[image]

closed 2022-12-28 by just-trey