just-trey/ankermake-m5-profile #5: Filename format contains comma which makes sending files to printer fail.

Vyvrnc, 2022-12-28:
When you try to send gcode from generated prusaslicer profile you will get print failed error. And it's because in the file name is comma in the layer height. Easy solution was to put dot instead. Could you please change the output filename format in the profiles to contain dot instead?

just-trey, 2022-12-28:
@Vyvrnc, what version of Prusa Sslicer are you using? Or are you using a Super Slicer? The comma should not appear in the file name as it is a parameter. When I generate a file using the output format it does not contain a comma.
example: {input_filename_base}_{digits(layer_height,1,2)}mm_{filament_type[0]}_{printer_model}.gcode
creates: 3DBenchy_0.20mm_PLA_M5.gcode

Vyvrnc, 2022-12-28:
I'm using PrusaSlicer 2.5.0 and it creates file:
Fab365_Foldable_StarWars_Millennium-Falcon_Cockpit_0,10mm_PLA_M5
or
jaw_0,10mm_PLA_M5
When I use Prusa Mini profile the output format is writen like this:
{input_filename_base}_{layer_height}mm_{initial_filament_type}_{printer_model}_{print_time}.gcode
then it's
jaw_0.1mm_PLA_MINI_1h4m

Vyvrnc, 2022-12-28:
@just-trey
Version: 2.5.0+win64
Build: PrusaSlicer-2.5.0+win64-202209060714

just-trey, 2022-12-28:
@Vyvrnc is the STL you are using free? If so can you provide a link?

Vyvrnc, 2022-12-29:
@just-trey The StarWars is but after registration. But the jaw file is part of this.
https://www.thingiverse.com/thing:4076827

just-trey, 2022-12-31:
Are you on a Mac by any chance?

PR #18 (2023-01-21): update read me and file name
Closes #5
commit f5ee193b (2023-01-21): update read me and file name (#18)

Closes #5
closed 2023-01-21 by just-trey