just-trey/ankermake-m5-profile #12: Q: Cura base plate obscuring underside

tomjn, 2023-01-06:
If I twist the view around in AnkerMake I can see the bottom of the print, but I can't do the same in Cura as it just shows the printbed model with the anker logo/grid. Is this expected?

I wonder if a version with only top faces might be better so it isn't rendered when viewed from underneath

gitsf, 2023-01-06:
That is how it appears for me as well. I too would like to be able to see the view of the model and not the print bed when viewing from a bottom angle.

just-trey, 2023-01-07:
@gitsf and @tomjn question for you. I want to keep the branding and look close. I made the png more transparent as a test. Would this meet the need? I know it is purely aesthetic. 

top
[image]
Bottom
[image]

tomjn, 2023-01-07:
That looks more usable, thanks for the change!

gitsf, 2023-01-08:
Yes that looks great.
Thanks

commit 200b4859 (2023-01-21): Q: Cura base plate obscuring underside
Fixes #12
closed 2023-01-21 by just-trey