just-trey/ankermake-m5-profile #19: Significand amount of filament shows up as "unknown" type when imported into AnkerMake slicer

chrivers, 2023-01-23:
I've tried to use PrusaSlicer to slice models for my Ankermake M5, using the profiles supplied here.

In the process, I've noticed that the following categories of gcode seem to register as "unknown" in AnkerMake:

 - Overhang perimeter
 - Top solid infill
 - Bridge infill
 - Skirt/Brim

For illustration, view in prusaslicer, with only the mentioned types showing:

[image]

View in AnkerMake, with *only* the "unknown" type showing:

[image]

Since AnkerMake is (so far) the easiest way to send jobs to the printer, it would be nice if the preview version was free of "unknown", although it doesn't seem to be a dealbreaker for printing.

I can help debug this further, if needed.

Xelinor, 2023-01-23:
This...isn't really a bug? Those are just features Ankermake slicer doesn't have/support. If it bothers you to see them as unknown, you can load the gcode from the device tab without looking at the preview and send them to the printer from there.

just-trey, 2023-01-23:
I actually do recommend what @Xelinor suggests as it will preview in that tab as well. I don't think it specificies line types in that preview screen. I do understand it would be nice to have but is outside of my control.

closed 2023-01-23 by just-trey

chrivers, 2023-01-23:
I looked at the generated G-code, and along with the explanation, that makes sense.

Seems AnkerMake just uses different `TYPE: foo` labels, and just happens to understand *some* of the same ones as PrusaSlicer generates... oh well :)